stanus74/home-assistant-saj-h2-modbus #1: Feature Request: Adding some exta registers

cebrox, 2024-06-13:
Hi,

can you add the register PV1Power (4073H), PV2Power (4076H), TotalPVPower (40A5H), Today_PVEnergy (40BFH), Month_PVEnergy (40C1H), Year_PVEnergy (40C3H) and Total_PVEnergy (40C5).

The PVEnergy-values are important for the energy-dashboard because the inverter_generation values includes the generation from the battery at night.

The PV1 and PV2 values are very interesting for me, because i have a east and west side string.

That would be really great!

Greetings
Cebrox

UDicke, 2024-06-13:
Hallo René, hast Du mal [https://github.com/dboeni/home-assistant-ampere-storage-pro-modbus](url) angeschaut? Ampere.Storage.Pro ist SAJ mit einem Label von EKD-Solar.

stanus74, 2024-06-13:
sensor.saj_total_power_generation, sensor.saj_inverter_year_generation, sensor.saj_inverter_month_generation,sensor.saj_power_generation_on_current_day is disable by default

you can activate these sensors under devices. go to filters and tick show deactivated

PV1Power and PV2Power have i added to my test system.

i will release a new version in a few days

UDicke, 2024-06-14:
I don't know how "saj_pv_power" is calculatet within the inverter.
In my plant the value of saj_pv_power does not return the sum of p1power and pv2power as I would expect. It differs sometimes significant from this sum. I believe that the reason for that is that I have additional panels with inverters in our plant. See this example:
[image]
My recommendation is to provide a calculatet "saj_total_pv_power" as a sum of pv1power and pv2power.

cebrox, 2024-06-14:
@stanus74 Thank you! You can close this issue after release :)
@UDicke Danke für den Tipp mit dem anderen custom_modul. Das der Ampere Storage nur ein umgelabelter SAJ ist, hatte ich auch schon rausgefunden und deswegen auch dieses Modul genutzt.

UDicke, 2024-06-14:
@cebrox hast Du denn auch zusätzliche Module? Sonst vermute ich es gibt gute Chancen, dass der Wert von  saj_pv_power bei Dir stimmt. Ich bin gespannt. 😊

stanus74, 2024-06-15:
weitere Sensoren wurden hinzugefügt

closed 2024-06-15 by stanus74