stanus74/home-assistant-saj-h2-modbus #4: Not available in integrations anymore?

Lars-, 2024-06-24:
Hi,

Since I've upgraded to 1.4.0, the integration doesn't show up in the new integrations modal anymore. When I downgraded to 1.3.0 it worked again. 

Maybe something worth looking at.

I will use 1.3.0 for now.

stanus74, 2024-06-25:
I cannot reproduce the error

I have done the following:
1. deleted the integration completely
2. installed version 1.3 
2. update to 1.4.

Integration is visible.

Try this:

Go to "Device and Service" > "ADD INTEGRATION" search for SAJ H2

Lars-, 2024-06-26:
I think it indeed had something to do with my setup. It seems to be updated to 1.4.0 automatically (even though hacs doesn't do that as far as I know) and now it works as it should.

Sorry for the unnecessary issue and have a nice day!

closed 2024-06-26 by Lars-