stanus74/home-assistant-saj-h2-modbus #14: Unable to see grid import real-time

tims04, 2024-10-19:
Thanks so much for your work on this, it has been a very interesting project for me to integrate this at home. 

I have a SAJ H2-5K-S2 Interter Plus SAJ BC2-HV1 battery.

I don't seem to be able to get the correct figure to measure the energy being imported (kWh) from grid in realtime?  Or perhaps I am missing something.

I am using sensor.saj_total_grid_power - should that show me what I am currently importing from the grid?

stanus74, 2024-10-27:
use this sensor for grid import/export data in realtime:  sensor.saj_grid_load_power

closed 2024-10-27 by stanus74

tims04, 2024-11-05:
Ok I think I have figured this out, grid import includes AC coupling from other independent inverter connected to the same panel (other inverter is not SAJ), hence I didn’t think I was seeing true grid import, because it includes import off other inverter.

UDicke, 2024-11-07:
> grid import includes AC coupling from other independent inverter connected to the same panel
@tims04 In such a case, you should use the data from the network operator's smart meter. We use the tibber pulse and before tibber I had a different read head on the meter from the grid. I would prefer this data anyway. If you have an additional inverter, you also have to consider the solar yield separately.