stanus74/home-assistant-saj-h2-modbus #20: Setup failed for custom integration 'saj_modbus': Unable to import component: No module named 'pymodbus.register_read_message'

jsimonic, 2025-01-04:
Hello,

after upgrade HA to 2025.1.0, saj modbus not working.

Error here:

Setup failed for custom integration 'saj_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/saj_modbus/__init__.py", line 7, in <module>
    from .hub import SAJModbusHub
  File "/config/custom_components/saj_modbus/hub.py", line 13, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

Please for help.
Thanks
Br,
Jernej

User3n, 2025-01-04:
Same had to rollback.

tims04, 2025-01-04:
Same issue here. Rolled back HA.

jsjhb, 2025-01-04:
There seems to be a deeper change in 2025.1 with respect to pymodbus.
A trivial change of commenting out the line where ReadHoldingRegistersResponse is called and the import from pymodbus.register_read_message did not solve the issue, in that case the integration could be started, but no modbus data retrieval was possible.

This seems to boil down to
`2025-01-04 08:45:26.333 WARNING (MainThread) [custom_components.saj_modbus.hub] Error during connection attempt: AsyncModbusTcpClient.__init__() got an unexpected keyword argument 'retry_on_empty'
  File "/config/custom_components/saj_modbus/hub.py", line 143, in try_read_registers
  File "/config/custom_components/saj_modbus/hub.py", line 118, in ensure_connection
  File "/config/custom_components/saj_modbus/hub.py", line 44, in _create_client

2025-01-04 08:45:26.336 ERROR (MainThread) [custom_components.saj_modbus.hub] Failed to reconnect Modbus client.

2025-01-04 08:45:26.336 ERROR (MainThread) [custom_components.saj_modbus.hub] Read attempt 2 failed at address 36608: Modbus Error: [Connection] Unable to establish connection
`

A quick glance at the ChangeLog did not give a hint to an updated version of pymodbus, but likely there was a jump from pre-3.7.0 to something newer.

stanus74, 2025-01-04:
I have just published a new release [v1.6.4](https://github.com/stanus74/home-assistant-saj-h2-modbus/releases/tag/v1.6.4) that should fix this issue. 
Please upgrade and let me know if you encounter any problems.

closed 2025-01-04 by stanus74

User3n, 2025-01-04:
All fine now

tims04, 2025-01-04:
Thank you, working perfectly now.

jsimonic, 2025-01-05:
Now work perfectly.
Thank you for quick fix this issue.

stanus74, 2025-01-05:
I am glad that it works
Thanks for the support