stanus74/home-assistant-saj-h2-modbus #22: Battery temperature

User3n, 2025-01-15:
The battery temperature shown are higher than the one shown in app by 4 deg C. Maybe each battery have  an temp censor? So what I see is another battery sensor?  Or do it need calibration?  I think the lowest battery is coldest and Saj need 22 deg to give full effect so it's important. I have  3 batteries.
And thanks for this lovely  add-on.

stanus74, 2025-01-17:
Thank you for your feedback and the great idea about the battery temperature sensors! Implementing this would require some additional effort, but I’ll see what I can do.

If you enjoy the project, you can find information in the README on how to support its development.

Thanks again and best regards

User3n, 2025-01-17:
I already bought you some coffee. But don't feel pressure just a suggestion if it's not too much work.

stanus74, 2025-01-18:
Thanks for that

new sensors will be added in one of the next versions
I am collecting suggestions

closed 2025-01-18 by stanus74