stanus74/home-assistant-saj-h2-modbus #24: Error reading modbus data

frencifrenci, 2025-02-05:
Hi,
after updating home assistant to version 2025.2.0, the component stopped working.

This is the error message:

custom_components/saj_modbus/hub.py:214

Error reading modbus data: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

How can it be solved?

Thanks

User3n, 2025-02-05:
Just to confirm, same here.

Regads

davidjbwatson, 2025-02-06:
updated core this morning now all entities unknown

hin402, 2025-02-06:
> just updated the core and all entites becom unknow.

rollback core to 2025.1.4 it back online

stanus74, 2025-02-06:
fixed with version v1.6.5 https://github.com/stanus74/home-assistant-saj-h2-modbus/releases/tag/v1.6.5
write if it works for you

frencifrenci, 2025-02-06:
It works!

Thank you very much!

And i buy for you a coffee!

User3n, 2025-02-06:
Works!

Thank you .

stanus74, 2025-02-08:
thanks for support / coffee