stanus74/home-assistant-saj-h2-modbus #27: No values for battey 2 and 3

User3n, 2025-02-21:
No values for battey 2 and 3, I have 3 batteries but only get values for battery one.
Again thanks for this great addon, without I knew nothing.

Battery 1 Temperature
23,0 °C
Battery 1 Voltage
316,5 V
Battery 1 Warning
0
Battery 2 Current
0,0 A
Battery 2 Cycle Count
0
Battery 2 Fault
0
Battery 2 SOC
0,0 %
Battery 2 SOH
0,0 %
Battery 2 Temperature
0,0 °C

stanus74, 2025-02-21:
also have 2 batteries and no values for battery 2 registers
But they are the registers according to the protocol (version 2022)

Perhaps this refers to the battery controller not battery

I wrote to SAJ a few weeks ago that I am a developer and asked for support.

Have not received any support yet, except that private developers cannot be supported due to privacy policy

stanus74, 2025-02-21:
https://saj-electric.com/contact

User3n, 2025-02-21:
Seems the page went down too many to complain :)

UDicke, 2025-02-21:
https://de.saj-electric.com/contact
info@saj-electric.com - da ist vermutlich der Verkauf
+49 162 2748767 - mit dieser Nr habe ich mal einen deutsch sprechenden Techniker von SAJ erreicht. Auskunft war mager. Die Anfrage betraf damals eine SerienNr (hat er abgefragt) von Ampere.Storage.Pro (EKD-Solar), evtl. war das der Grund für "keine Auskunft"...

stanus74, 2025-02-22:
I have already asked several times if I can get access to current register overviews and technical information, but have not received a reply for several weeks.

As one of the few developers for the SAJ H2 in Home Assistant, it is disappointing that SAJ does not provide any support to the community. This does not reflect well on the company, especially when compared to other inverter manufacturers who are actively working with developers.

If anything changes or I do receive official information, I will of course let you know here.

closed 2025-02-24 by stanus74