stanus74/home-assistant-saj-h2-modbus #29: Functions stopped working for 1.80

odiechoo, 2025-03-04:
Firstly, thanks for the continued great work!

Upon upgrade to 1.8, all the new battery information went unavailable. The charging control became unavailable as well. When we try to toggle the charging control, the entire modbua became unavailable. 

I had to do a restore to recover back to version 
1.7 to get it all working again.

odiechoo, 2025-03-04:
[home-assistant_2025-03-04T21-21-08.876Z.log](https://github.com/user-attachments/files/19077841/home-assistant_2025-03-04T21-21-08.876Z.log)

Here it is. I'm hoping that the upload of the log works....

Mr-Alicates, 2025-03-05:
Hello, I see these errors too, However I have a H1 instead of an H2, so take these logs with a pinch of salt:

`2025-03-05 22:54:34.127 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:54:34.328 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 13895: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:54:36.330 ERROR (MainThread) [custom_components.saj_modbus.hub] Error reading charging state: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:54:50.477 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16695: Modbus Error: [Input/Output] Invalid response from address 16695
2025-03-05 22:54:52.679 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:54:52.880 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16433: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:54:54.881 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:54:55.082 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 40960: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:54:57.084 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:54:57.285 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 13830: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:54:59.286 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:54:59.487 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 13895: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:55:01.488 ERROR (MainThread) [custom_components.saj_modbus.hub] Error reading charging state: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:15.341 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16695: Modbus Error: [Input/Output] Invalid response from address 16695
2025-03-05 22:55:17.542 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:17.744 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16433: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:55:19.745 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:19.946 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 40960: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:55:21.947 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:22.147 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 13830: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:55:24.150 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:24.351 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 13895: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:55:26.353 ERROR (MainThread) [custom_components.saj_modbus.hub] Error reading charging state: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:40.219 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16695: Modbus Error: [Input/Output] Invalid response from address 16695
2025-03-05 22:55:42.421 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:42.622 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16433: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 22:55:44.624 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 22:55:44.824 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 40960: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
[... cut ...]
2025-03-05 23:00:34.895 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16433: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]
2025-03-05 23:00:36.897 ERROR (MainThread) [custom_components.saj_modbus.modbus_data_readers] Error reading modbus data: 'AsyncModbusTcpClient' object has no attribute 'host'
2025-03-05 23:00:37.098 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 40960: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient XX.XX.XX.163:502]`

stanus74, 2025-03-06:
please downgrade to version 1.7.

under certain conditions when using the switch, the modbus is overloaded

a new version will be released at the end of this week

jsjhb, 2025-03-09:
I have seen similar behavior of SAJ's AIO3 module when there is more the on party communicating with the AIP via Modbus TCP.
So it is most probably important to have some kind of locking in place to not talk simultaneously with other ongoing traffic. This is impossible when using other management devices, but the integration itself should be aware of current operation.

stanus74, 2025-03-09:
My integration doesn't prevent other devices from accessing the adapter over the network. Instead, it features comprehensive connection checks and smart retry logic to handle overload and communication issues gracefully.

Version 1.8.1 is out

closed 2025-03-09 by stanus74

odiechoo, 2025-03-09:
Thank you... its tested working. There are several of these errors, but it doesn't interfere with the functionality of the integration.

2025-03-09 20:01:51.471 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-09 20:01:51.472 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16695: Modbus Error: [Input/Output] Invalid response from address 16695
2025-03-09 20:01:53.474 ERROR (MainThread) [custom_components.saj_modbus.modbus_readers] Error reading modbus data: Reconnection needed due to read failure.
2025-03-09 20:02:28.520 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-09 20:02:28.520 ERROR (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 failed at address 16695: Modbus Error: [Input/Output] Invalid response from address 16695
2025-03-09 20:02:30.521 ERROR (MainThread) [custom_components.saj_modbus.modbus_readers] Error reading modbus data: Reconnection needed due to read failure.

stanus74, 2025-03-09:
does your inverter have the latest firmware. previously the connection via AIO3 was not stable.

Sometimes restarting the AIO3 with the SAJ O&M app or directly on the AIO3 via the menu also helps

> Since the Firmware Update for H2 Inverter (Version Module V1.211 , Displ. Board V1.070, Contr.Board V4.066) the conncection quality is improved. I had no connection errors since the update at 14 Nov. 2024.

odiechoo, 2025-03-09:
Yes, I think I have the latest. The numbers above seems the wrong way around. Anyway, my Display Board is v5.010 and my control board is v2.011