stanus74/home-assistant-saj-h2-modbus #30: Optimizing Home Battery and EV Charging with Home Assistant and Tibber

Hadziagovic, 2025-03-04:
First, I want to thank you for the fantastic job you are doing!

I am new to Home Assistant and would appreciate a feature that can stop the discharge of the home battery when the EV is charging. My electricity provider, Tibber, has a feature called smart charging that plans and schedules the charging of the electric car when electricity is cheapest, usually at night. I have also activated Grid Rewards, which is a support service for the power grid, and the charging schedule can change at any time. For example, if Grid Reward is active during the night when the solar panels are not producing electricity, the EV charges from the home battery first, and in this way, it does not support the grid, which I want to avoid.

I am still waiting for the API key to integrate the Charge Amps Halo charging box into Home Assistant. Once the integration is complete, the status of the charging box can be used as a trigger. For example, when the charging of the EV starts, the operating mode of the inverter is memorized, and the discharge of the home battery is stopped. When the charging of the electric car stops, the operating mode of the inverter is restored to the mode that was before.

stanus74, 2025-03-05:
I am happy if my integration is helpful

I have already asked SAJ Germany several times about current Modbus protocols and support for the further development of my integration.

SAJ does not want to support private developers

Unfortunately, I can't help you with your problem.

I would like to expand my integration with even more functions.

You are welcome to contact SAJ and ask why developers are not supported on Github.

service.de@saj-electric.com
Richard Zhang
Technical Engineer

Translated with DeepL.com (free version)

UDicke, 2025-03-05:
> would appreciate a feature that can stop the discharge of the home battery when the EV is charging

you can do this by using evcc (the big solution which I have installed for this and mainly for a brilliant surplus charging) or set the relevant modbus register using the [HA modbus integration](https://homeassistant.local:8123/config/integrations/integration/modbus) by yourself. The registers to set depend on the "working mode" (Register 13895) you are using.

My normal Working mode is "Backup Mode"

[image]

and the register 13905 can be set to 100% for stopping the use of the battery

[image]

To get back to "use the Battery" this register has to be set to your "backup SOC" i.e. 20%

If your normal working mode is "Self use mode" the register to use is 
Register 13892 (BatOnGridDisDepth - Battery on grid discharge Depth) instead of 13905 and the default value of this register is 10%

... or @stanus74 decides to implement this later...  😊

Hadziagovic, 2025-03-06:
Thanks for the tip, @UDicke! EVCC looks promising, but unfortunately, my charging box isn’t on the list. I have a Charge Amps Halo.

Charge Amps is migrating its communication protocol from CAPI (Charge Point API) to OCPP (Open Charge Point Protocol), so that might open up the possibility of integrating Halo into EVCC.

My next question is how to install the HA Modbus integration that you referred to, as the link you shared doesn’t seem to work.

I am also considering using an energy meter for the charging box and using the current as a trigger instead of integrating the charging box.

UDicke, 2025-03-06:
sorry for the wrong link!
https://www.home-assistant.io/integrations/modbus

and you will write from HA directly to your SAJ-H2. I did this already and I am still doing this beside evcc, i.e. setting register 13905 to 100% after charging from grid depending on the price. It does not make sense to charge from grid for i.e. 30,5 Cent and one hour later discharge the battery while the price is 30,8 Cent, so I disable the use of the battery until the price is higher.

If you want to check your inverter via modbus directy I recommend qmodmaster, i.e. checking your working mode...
https://sourceforge.net/projects/qmodmaster/files/latest/download

If there are any questions you can ask via [mail](mailto:u.dicke@gmx.de)

stanus74, 2025-03-06:
since version 1.7. there is a switch in my integratin to change the inverter mode from “Time-of-Use <> Self Consoumption” mode

In “Time-of-Use” mode, the battery is not discharged.

Simply set the charging power to 1% during the time the car is charging, and the power will only be drawn from the grid and not from the battery.

stanus74, 2025-03-06:
I have not yet used the backup mode myself.

As already mentioned, I receive no support from SAJ at all.
Even after my request, I have not received an updated Modbus protocol.

I would like to integrate new functions, but I need precise information to ensure that everything runs safely for users and does not cause any issues or damage.

A new version with the ability to set charging times will be released this weekend.
Version 1.8 had to be temporarily removed.

jsjhb, 2025-03-09:
I have seen similar functionality in Energiekonzepte Deutschland's (ekd) rebranded version for the HS2 - AmpereStorage.Pro - with their management appliance Ampere.IQ:
Grid charging is done via passive mode, described in #18. If the AppMode is set to "3" (passive mode) and the BatOnGridDisDepth is set to 100 or at least higher than the current SOC the system does not discharge - this would be the time for battery-independent charging of the car.

I have plans to implement this here, but ATM my Ampere.StoragePro is broken and ekd is giving a fscking damn about it.
I also acquired a separate HS2, but have to set it up and will then commence testing.

stanus74, 2025-03-09:
I have tested the approach suggested by the user – it appears to work. However, due to the lack of documentation in the Modbus protocol and from SAJ’s side, I will not incorporate this functionality into the integration.

This integration is solely intended for reading registers. Since version 1.8, you can switch the working mode to "Time-of-Use" to prevent battery discharge.

I welcome collaborative development – feel free to extend the integration as you see fit.

stanus74, 2025-03-09:
does your inverter have the latest firmware. previously the connection via AIO3 was not stable.

Sometimes restarting the AIO3 with the SAJ O&M app or directly on the AIO3 via the menu also helps

> **Since the Firmware Update for H2 Inverter (Version Module V1.211 , Displ. Board V1.070, Contr.Board V4.066) the conncection quality is improved. I had no connection errors since the update at 14 Nov. 2024.**

Hadziagovic, 2025-03-09:
When using the "Time of Use" mode and setting battery charging to 0%, the battery charges at 100 W. If the battery charging is set to 0%, it should logically not charge at all, meaning 0 W instead of 100 W. If it still charges at 100 W, there might be a system setting or a safety minimum causing a small amount of charging to occur.

stanus74, 2025-03-09:
I have just set the charge percentage to 0% and it was transferred correctly to the inverter

Did you wait for an interval?

look in the SAJ O&M or app under Working Mode to see what it says there

[image]

Hadziagovic, 2025-03-09:
Right now, my battery is charging from the solar panels. I'll check and get back tonight.

UDicke, 2025-03-09:
I would like to mention the [extended support](https://docs.evcc.io/en/docs/features/battery#active-battery-control) for the SAJ-H2 (Ampere.StoragePro) within evcc again, because my experience with this is very good. I use it since it was implemented including charging from grid with Tibber and "disable discharging" during loading the car from grid. Currently homebattery "charging from grid" still need some manual steering for optimal results, as "plans for homebattery" do not exist yet.

The handling of this within evcc is simple and does only use 
Register 13905 for "disable/enable usage of battery". This requires register 13895 is set to 2 (AppMode = 2 = backup mode).
The SOC levels can be configured, default 20% backup and 95% disable value (I modified to 12% and 100%)
Register 13895 (AppMode) is switched to 1 (AppMode = 1 = time mode) for "loading home battery from grid".
In addition register 13828 (Charge time enable control) is set to "1" for use of the "first charge time" and
register 13830 (First charge start time) is set to "0" (that is 00:00 h)
register 13881 (First charge end time) is set to "173B" Hex (that is 23:59 h)
register 13832 (First charge power time) is set to "7F64" Hex (means alls day of the week and 100% power)
Documentation:

[image]

my values:

[image]

In my opinion I can leave the loading power with 100%, because the saj inverter regulates the charging power according to certain parameters such as temperature, SOC and magic 😊 or whatever. When loading from grid during a cheap period it seem good to load with max. power
This is what I observed when loading from 13:00 to 14:00 h on 1.March 25:

[image]

closed 2025-03-11 by stanus74