stanus74/home-assistant-saj-h2-modbus #37: Recommended Visualization Tool: Sunsynk Power Flow Card

stanus74, 2025-03-15:
For users of my integration, I highly recommend the Sunsynk Power Flow Card (available via HACS) to beautifully visualize your energy production, consumption, battery status, and individual load details. This intuitive Lovelace card provides a clear and attractive overview of your entire energy system, allowing you to monitor and optimize your power usage at a glance.

Check it out here:
https://github.com/slipx06/sunsynk-power-flow-card

[image]

Yaml code which I currently use

```yaml

type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 10000
  shutdown_soc: 20
  show_daily: true
  max_power: 9000
  soc_end_of_charge: 90
  shutdown_soc_offgrid: 20
solar:
  show_daily: true
  mppts: 1
load:
  colour: "#5fb6ad"
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  show_absolute_aux: false
  aux_loads: 2
  aux_load1_name: Boiler
  aux_load2_name: HK IR
  aux_load1_icon: mdi:water-boiler
  aux_load2_icon: mdi:heat-wave
  animation_speed: 8
  max_power: 8000
  auto_scale: true
  dynamic_colour: true
  path_theshold: 100
  off_threshold: 0
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.sunsynk_inverter_voltage
  load_frequency_192: sensor.saj_r_phase_grid_frequency
  inverter_current_164: sensor.
  inverter_power_175: sensor.saj_inverter_power
  grid_connected_status_194: null
  inverter_status_59: sensor.saj_inverter_status
  day_battery_charge_70: sensor.saj_battery_today_charge
  day_battery_discharge_71: sensor.saj_battery_today_discharge
  battery_voltage_183: sensor.saj_battery_1_voltage
  battery_soc_184: sensor.saj_battery_1_soc
  battery_power_190: sensor.saj_battery_power
  battery_current_191: sensor.saj_battery_1_current
  battery_temp_182: sensor.saj_battery_temperature
  grid_power_169: sensor.saj_grid_load_power
  day_grid_import_76: sensor.saj_sum_all_phases_feed_in_today
  day_grid_export_77: sensor.saj_sum_all_phases_sell_today
  grid_ct_power_172: sensor.saj_ct_grid_power_watt
  day_load_energy_84: sensor.saj_total_today_load
  essential_power: none
  nonessential_power: none
  aux_power_166: none
  day_pv_energy_108: sensor.saj_power_current_day
  pv1_power_186: sensor.saj_pv1_power
  pv2_power_187: null
  pv1_voltage_109: sensor.saj_pv1_voltage
  pv1_current_110: sensor.saj_pv1_total_current
  radiator_temp_91: sensor.saj_inverter_temperature
  environment_temp: sensor.gw1100a_outdoor_temperature
  aux_load1: sensor.sd_1_power
  aux_load2: sensor.sd_hk_3_power
  
inverter:
  three_phase: true
  modern: true
  autarky: power

```

bierma1966, 2025-03-15:
Thx stanus for New Update.
I try to install it tomorow

User3n, 2025-03-17:
Nice Thx.

Deapper, 2025-09-08:
> Thanks for making this beautiful card.
> 
> I am using a SAJ HS210KW inverter and am using SAJ H2 Inverter Modbus integration. The grid load is always shown as 0 even when we are pulling from grid.
> 
> <img alt="Image" width="520" height="441" src="https://private-user-images.githubusercontent.com/47166657/486565307-cba83975-1f99-4b0c-9aae-51e97c3a2c87.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTczMTI4NDMsIm5iZiI6MTc1NzMxMjU0MywicGF0aCI6Ii80NzE2NjY1Ny80ODY1NjUzMDctY2JhODM5NzUtMWY5OS00YjBjLTlhYWUtNTFlOTdjM2EyYzg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA4VDA2MjIyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiMjgzZThkZDc4ZDQ1NTZjYzhjNDRlMDQ2N2UxNDgzZTRlNTM2Y2I3MWY5YTI1M2NjODllMTM3NzRlN2E5YjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0v3dDXqIgTwq7w35-dotPyAgxxgG09rVDLjydfIYpF4">

Have you checked the states in developer tools?
day_grid_import_76: sensor.saj_sum_all_phases_feed_in_today
  day_grid_export_77: sensor.saj_sum_all_phases_sell_today
  grid_ct_power_172: sensor.saj_ct_grid_power_watt

tsvirk, 2025-09-08:
> Have you checked the states in developer tools? day_grid_import_76: sensor.saj_sum_all_phases_feed_in_today day_grid_export_77: sensor.saj_sum_all_phases_sell_today grid_ct_power_172: [...]

Thanks for pointing in right direction, some of the entities were disabled and had to replace a few to get this working. The modified entities can be seen in below screenshot.

[image]