stanus74/home-assistant-saj-h2-modbus #43: discharge lock for the battery via the integration

UDicke, 2025-04-12:
Please implement the option to switch a discharge lock for the battery via the integration. In backup mode, the current value of register 3651H should be saved and then set to 100%. To disable the discharge lock, the register must then be reset to the saved value.
In self-use mode, the register to be used is 3644H.

stanus74, 2025-04-14:
thank you for your suggestion and your kind words! 😊
I'm always happy to receive ideas and feedback from users.

Several users have already asked for an option to discharge the battery via Home Assistant, and I’m happy to say that this will be part of the next version of the integration.

After that, I’ll gladly look into your request and try to include it in one of the following updates. 

Thanks again for your support!

stanus74, 2025-04-18:
Register 3644 is not writable with the AIO3. Since many people use this, it makes little sense to implement it.

There is a simple way to avoid discharging
Simply switch to time-of-use mode with the "SAJ Charge Control" switch, set time 0:00 - 23:59, power 0 

In time-of-use mode, the battery is not discharged

closed 2025-04-18 by stanus74

UDicke, 2025-04-18:
Thank you for this alternative solution for a discharge lock! 👍