stanus74/home-assistant-saj-h2-modbus #46: Error reading modbus data: Reconnection needed after 3 failed read attempts at address xxxxx, unit 1. Last error: Modbus Error: [Input/Output] Invalid or error response

GHRRGH, 2025-04-18:
Hello, 

I was able to successfully add the integration with the default settings and read out the resulting data.
Afterwards I did enable a couple of entities (and disabled others) based on need. Primarily the total power for use in the energy dashboard, which didn't seem to be activated by default.

A bit later, almost all entities have gone to "Unknown" and the logs are flooded with the below message:

> Error reading modbus data: Reconnection needed after 3 failed read attempts at address xxxxx, unit 1. Last error: Modbus Error: [Input/Output] Invalid or error response

Any idea what could be the cause ? 

I'm running Home Assistant 2025.4.2 on Docker.

stanus74, 2025-04-18:
Which Version 2.0.0? 
Do you use AIO3? Which Firmware Version? 1.211?

GHRRGH, 2025-04-18:
I'm on v2.0.0 indeed.

AIO3 indeed.
Firmware version seems to be 1.202

I had the scan interval at 60, but updated it to 600 now to see if that helped (not so far)

stanus74, 2025-04-18:
you must restart AIO3. Use the SAJ O&M App or the Button on AIO3 . You can go to the Menu for Restart AIO3

did you already have the previous versions installed?
or did you just install the integration today?

GHRRGH, 2025-04-18:
I have restarted it already, via the device itself (to put it in the default VLAN to make sure it wasn't a network problem)

I installed it today indeed and it's the first version I've used.

stanus74, 2025-04-18:
again.
At the beginning, data from the sensors was displayed and after a while they all went to ‘unknown’ - is that correct?
Then reading via port 502 Modbus works correctly

Do the following:
Downgrade to version 1.9.0 (there is a larger pause when reading the register blocks)

and test it and let us know

GHRRGH, 2025-04-18:
I had already commented out some of the reader blocks and after the reboot they did seem to come back up.
I'll reduce the update intervals again and keep you posted!

[image]

stanus74, 2025-04-18:
thats not the problem
comment below on the command from
#await asyncio.sleep(2)

and insert the value 1, i.e. like this
await asyncio.sleep(1)

[image]

GHRRGH, 2025-04-18:
It seemed to work with a limited number of reader functions at a scan frequency of 600
I've now re-enabled the other reader functions with a 1 second sleep at a scan frequency of 300.
==> the sensors seem to stay (not reverting to unknown). However, the logs fill up with errors again. 

Will check and try to find the sweet spot

stanus74, 2025-04-18:
downgrade to version 1.9. and test it and write me.

stanus74, 2025-04-18:
check your inverter firmware . when not have the version below you need a upgrade, write SAJ
from the readme.md  (version November 2024) 

> **Since the Firmware Update for H2 Inverter (Version Module V1.211 , Displ. Board V1.070, Contr.Board V4.066) the conncection quality is improved. I had no connection errors since the update at 14 Nov. 2024.**

GHRRGH, 2025-04-21:
> downgrade to version 1.9. and test it and write me.

The downgrade did not help unfortunately. With some of the readers commented out it works a couple of hours, but eventually fails as well.

> check your inverter firmware . when not have the version below you need a upgrade, write SAJ
from the readme.md (version November 2024)

I'm checking how to get the firmware upgrade. Will let you know once I have some results!

GHRRGH, 2025-04-21:
@UDicke It's only 1 inverter and I only use Modbus for this. I do notice that the cloud updates are also sometimes delayed.
For now, I've commented out all but the 4 entities I need, which seems to hold so far. 

It also seems to fail upon restart of HA. Possibly due to connections still being open. (It requires a restart of the ethernet connector to function again)

UDicke, 2025-04-21:
@GHRRGH are you using LAN or W-Lan? My connection via LAN is very stable. (Ampere.Storage.Pro with external AOI3 Modul)

GHRRGH, 2025-04-21:
@UDicke LAN as well. (H2 with the AOI3 Module)

stanus74, 2025-04-21:
Have you checked the firmware version? 
Older firmware the Mobus connection was bad with blocking ...
Write SAJ to get the current firmware inverter
AIO3 1.211, Displ. board 1.070, Contr. board 4.066

GHRRGH, 2025-04-21:
I have, I'm on an older version indeed:
Module firmware: V1.202
Display board: V1.030
Control board: V1.009

I'm checking how to get it updated.

closed 2025-04-21 by stanus74