stanus74/home-assistant-saj-h2-modbus #49: Integration are not available on HACS

marcosturiale, 2025-04-24:
Hi,
First of all I want to thank you for your work. The integration works perfectly and you saved my life!

I would like to point out that the integration is not available for direct download using HACS (using searche function). 
It seems like it conflicts with another integration (https://github.com/wimb0/home-assistant-saj-r5-modbus).

I installed it anyway by adding a repository manually in HACS.
Could you please check?

Thank you

stanus74, 2025-04-24:
thank you for the info
i have create a beta version to change the domain to "saj_h2_modbus"v
more here https://github.com/stanus74/home-assistant-saj-h2-modbus/releases/tag/v2.0.2-beta

irishwoody, 2025-04-24:
Hello - Many Thanks 

I've just connected to a SAJ H3S system and seem to be getting readings on a good number of entities !

I'm happy to assist to test against H3S system if that's is of interest ?

Woody

closed 2025-04-26 by stanus74