stanus74/home-assistant-saj-h2-modbus #50: SAJ H3S Compatibility?

stanus74, 2025-04-24:
> Hello - Many Thanks 
> 
> I've just connected to a SAJ H3S system and seem to be getting readings on a good number of entities !
> 
> I'm happy to assist to test against H3S system if that's is of interest ?
> 
> Woody 

 _Originally posted by @irishwoody in [#49](https://github.com/stanus74/home-assistant-saj-h2-modbus/issues/49#issuecomment-2828384915)_

irishwoody, 2025-04-25:
Hi,

I have a single phase SAJ   HS3-6K-S2-W-PX-IE  "All in one energy System"

I played around enabling and disabling the entities in the Integration - the ones I have shown below seem to contain valid data from the Modbus interface.  I didn't try writing to the charge control as not 100% clear if this would poke data into a undefined interface or locations resulting in possible loss of function.  Seems most sensors tied to any specific battery number do not work, but the unnumbered battery sensors work, as does first 2 PVs, the first Phase "R phase" and the Energy / Loading etc.  I have matched a good number of these sensors with the data presented by the elekeeper App.

Yesterday - I had checked to see if Port 502 was open and Nmap showed me it was Open with a "mbap" service.  Then I tried your integration.  Many thanks for the Integration as the SAJ esolar one stopped working last week due to the new cloud platform rollout and old platform being switched off.  In anycase - I much prefer the local network connection and it also has an increased number of working sensors.  If I can be of any help let me know ?

Woody

[image]

[image]

[image]

irishwoody, 2025-04-25:
It would be nice to get the power factor as a graph / type measurement ; the default display is not so helpful. As a factor it has no units of measurement - dont know if that is issue or is it a definition / class issue

[image]

stanus74, 2025-04-25:
the sensors look good so far, including the times. Did you set the time-of-use mode? App Mode 1 is TOU
Try changing the working mode in the SAJ app and the times etc..

I will change the type measurement in the next version