stanus74/home-assistant-saj-h2-modbus #54: Upgrade and historical data

toast1234, 2025-04-27:
The upgrade procedure does not mention if the recorded data is migrated to the new integration. Does the entity name change with the new integration? E.g mine are prefixed saj_mbus_.  A backup is of course always good to take but can you please clarify in the procedure if historical data is kept, migrated or lost as a part of the process?

stanus74, 2025-04-27:
in the input field you should use the same prefix as you used during the last installation

if you did not use "saj" as presented, then enter what you used here

[image]

toast1234, 2025-04-27:
Thats great
Thanks

closed 2025-05-13 by stanus74