stanus74/home-assistant-saj-h2-modbus #61: How to Schedule Battery Charging and Discharging from the Grid

stanus74, 2025-05-05:
## 🔋 How to Schedule Battery Charging and Discharging from the Grid 
### 🔌 Programming Battery Charging

1. **Set Charge Day Mask (Input)**
   - Make sure all weekdays are selected.  
     ➤ Value `127` means all days are enabled.  
     ➤ This is a **bitmask**:  
       - Monday = Bit 0 = 1  
       - Tuesday = Bit 1 = 2  
       - Wednesday = Bit 2 = 4, etc.

2. **Set Charging Times**
   - Configure:
     - `Charge Start Time (Time)`
     - `Charge End Time (Time)`

3. **Set Charging Power**
   - Set the `SAJ Charge Power Percent (Input)` to the desired value.  
     ➤ Example: If the inverter has 8 kW max and you set 10%, it will charge with 800 W.

4. **Activate Charging Control**
   - Turn on the **Charging Control Switch**.
   - 🔍 Check the SAJ app initially to verify that charging starts correctly.

---

### ⚡ Programming Battery Discharging

1. **Same as Charging**
   - Define start and end time, power percentage, and day mask just like for charging.

2. **Activate Discharging Control**
   - Turn on the **Discharging Control Switch**.

3. **Enable Time Slots**
   - If you defined multiple discharging time slots (e.g., slot 1 and 2),  
     set the `SAJ Discharge Time Enable (Input)` to the **sum of active bits**:  
     ➤ Slot 1 (Bit 0) = 1  
     ➤ Slot 2 (Bit 1) = 2  
     ➤ Enable both = `1 + 2 = 3`

4. **Initial App Configuration**
   - ⚠️ On first use, **open the SAJ App**, go to **Working Mode**, and enable  
     **"Battery Export to the Grid"** to allow discharging.  
     ➤ See the README for more details.

### ⚠️ Important Notes

- Always follow the **correct order** of steps.
- When activating **Discharging Control**, **only the first time slot** is used unless the `Discharge Time Enable` input is set correctly.
- When you disable either **Charging** or **Discharging Control**, the system:
  - Deactivates all time slots (sets the value to 0)
  - Resets the inverter mode to **Self Consumption**

bierma1966, 2025-08-26:
Quick solution.
Actived discharging in the moring with 1% and setting time slot from 6 to 11 am.

Deapper, 2025-08-26:
> Quick solution. Actived discharging in the moring with 1% and setting time slot from 6 to 11 am.

Thanks for the suggestion. And this does work sort of. Problem with this idea is if load goes over solar output the difference comes from the grid

Deapper, 2025-08-27:
> Play with Discharging value, it must be a little bit higher than your
> house consumption .

If one could only predict ones wife's energy consumption

stanus74, 2025-08-27:
Use number.saj_battery_capacity_charge_upper_limit_input

Set it to actually value. this stop the battery charging

you can also create a automation with  time base

Deapper, 2025-08-30:
Thanks that worked a treat

Deapper, 2025-08-31:
Please one more thing. How do you instantly discharge? I'd like to setup automation to discharge if soc is above 75%.

I can only see timer discharge. Thank you

etrafiee, 2025-10-07:
Hi @stanus74,

thanks for your add on.
I don't seem to be able to configure multiple time slots for discharging.
I've tried the Lovelace card and manual inputs to the fields.
no matter what, it only picks up the first slot.

any insights on that?

stanus74, 2025-10-08:
was fixed with version 2.6.1 
https://github.com/stanus74/home-assistant-saj-h2-modbus/releases/tag/v2.6.1

quazzie, 2025-10-30:
It seems i can't choose over night times, ex. 20:00 -> 07:00 it stops charging but if i use 20:00 -> 23:59 it charges. Am i missing something ?