stanus74/home-assistant-saj-h2-modbus #74: Missing input_number mappings in __init__.py ?

quazzie, 2025-05-19:
### Inverter model

SAJ HS2 10

### Inverter firmware version

v1.212

### Which Modbus TCP converter are you using?

AIO3

### Modbus converter firmware version

v4.084

### Describe the issue

I think the new v2.2.3 input things are missing in `__init__.py` ?

There is no mapping in the log files and nothing happens when i change the configuration setup `input_number.saj_battery_on_grid_discharge_depth`

There are mappings for some.

Or am i missing something ?

Btw thanks for your work!

### Home Assistant version

2025.5.2

### Log output

2025-05-19 20:23:48.343 INFO (MainThread) [custom_components.saj_h2_modbus] Found input_number entity: input_number.saj_charge_day_mask
2025-05-19 20:23:48.344 INFO (MainThread) [custom_components.saj_h2_modbus] Found input_number entity: input_number.saj_charge_power_percent
2025-05-19 20:23:48.344 INFO (MainThread) [custom_components.saj_h2_modbus] Found input_number entity: input_number.saj_discharge_day_mask
2025-05-19 20:23:48.344 INFO (MainThread) [custom_components.saj_h2_modbus] Found input_number entity: input_number.saj_discharge_power_percent
2025-05-19 20:23:48.344 INFO (MainThread) [custom_components.saj_h2_modbus] Found input_number entity: input_number.saj_export_limit
2025-05-19 20:23:48.344 INFO (MainThread) [custom_components.saj_h2_modbus] Found input_number entity: input_number.saj_app_mode

stanus74, 2025-05-19:
Hi! Thanks a lot for your feedback 😊

As of version **v2.2.3**, you no longer need to manually create or use `input_number.*` helpers. The integration now includes its own `NumberEntity` instances (e.g., `number.saj_battery_on_grid_discharge_depth_input`) that communicate directly with the inverter.

👉 **Please use the `number.*` entities only**, and remove any related `input_number.*` entries from your `configuration.yaml`.

The old mapping log messages were removed because they’re no longer necessary.

📄 I’ve also removed the YAML example for `input_number:` from the README to avoid confusion.

Thanks again for testing — feel free to reach out with any more questions! 😊

quazzie, 2025-05-19:
Great, i see the number entities now. I was more confused as they are not connected to the SAJ device so i did not see them when i looked at the device. But if i search for them i can find them.

closed 2025-05-19 by quazzie

quazzie, 2025-05-19:
Btw the old logs are not removed, i still get logs for input_number mappings :) but as i removed them i get "Please add this to configuration.yaml" ...

:)

stanus74, 2025-05-19:
this is only a warning. it will remove in the next version in few days

stanus74, 2025-05-19:
do you know my lovelace card for SAJ Integration 
https://github.com/stanus74/saj-h2-lovelace-card

quazzie, 2025-05-19:
Yes, but i do not schedule so i have not had a use for it at the moment.

quazzie, 2025-05-20:
Btw, is there no 'SAJ Import Limit' ?

stanus74, 2025-05-20:
see changelog https://github.com/stanus74/home-assistant-saj-h2-modbus/blob/main/CHANGELOG.md

more description than the name and register i don't have from saj