stanus74/home-assistant-saj-h2-modbus #75: v2.2.2 regression: sensor.saj_power_current_day lacks native_last_reset and causes negative midnight values in Energy Dashboard

Caslinovich, 2025-05-23:
### Inverter model

SAJ H2-10K-T2

### Inverter firmware version

1.211

### Which Modbus TCP converter are you using?

AIO3

### Modbus converter firmware version

AIO3 FW 4.084

### Describe the issue

Hi, thank you for maintaining this integration!

After updating to version 2.2.2, I noticed that the `sensor.saj_power_current_day` no longer behaves correctly in the Home Assistant Energy Dashboard.

In the changelog, it's stated that `state_class` was changed from `total_increasing` to `total` and that `native_last_reset` was added to help HA correctly interpret daily resets.

However, **in my system the sensor does not expose `native_last_reset`** or any reset information at all. It only shows:

state_class: total
unit_of_measurement: kWh
device_class: energy
icon: mdi:solar-power
friendly_name: SAJ Power current day

As a result, Home Assistant interprets the reset at midnight as a negative spike.

This regression only started after updating to 2.2.2. With version 2.2.1, the sensor had state_class: total_increasing and the Energy Dashboard worked correctly.

### Home Assistant version

2025.5.0

### Log output

No errors in Home Assistant logs.

The issue is with the behavior of the following daily-resetting sensors:

- `sensor.saj_power_current_day`

[image]

- `sensor.saj_sum_all_phases_sell_today`
- `sensor.saj_battery_today_charge`
- `sensor.saj_sum_all_phases_feed_in_today`

All of them reset to 0 at midnight, but since `state_class` is now set to `total` (and not `total_increasing`), and they do not expose `native_last_reset`, Home Assistant misinterprets the reset as a large negative drop.

This results in incorrect negative values between 00:00 and 01:00 in the Energy Dashboard — for example, showing **-13.5 kWh of solar production** after midnight.

This behavior did not happen in v2.2.1, where `state_class` was `total_increasing` and handled properly.

stanus74, 2025-05-23:
Hi, thanks a lot for your detailed explanation and for testing the update!

You're right that these sensors — like `sensor.saj_power_current_day`, `sensor.saj_battery_today_charge`, etc. — reset daily and therefore have `state_class: total` **by design**. Since they don’t track historical increases, they are **not compatible** with the Home Assistant Energy Dashboard.

Instead, for proper long-term tracking in the Energy Dashboard, you should use the **total-accumulating sensors** provided by the integration:

### ✅ Use these sensors in the Energy Dashboard:

**Electricity grid:**

* `sensor.saj_sum_all_phases_feed_in_total` (Grid consumption)
* `sensor.saj_sum_all_phases_sell_total` (Grid export)

**Solar panels:**

* `sensor.saj_total_power_generation` (Total solar production)

**Home battery:**

* `sensor.saj_battery_total_discharge` (Battery discharge)
* `sensor.saj_battery_total_charge` (Battery charge)

You can configure these in:
**Settings → Dashboards → Energy → Edit → Assign sources**

---

### ℹ️ Note:

If some sensors are missing from the dropdown:

* Go to **Settings → Devices & Services → Entities**
* Filter for **Disabled** status
* Enable the required sensors

---

Full Guide https://github.com/stanus74/home-assistant-saj-h2-modbus/discussions/73

I have been using these sensors myself for many months now

Thanks again for reaching out! Let me know if you'd like help mapping the sensors further. 😊

Caslinovich, 2025-05-23:
Thank you so much for your detailed explanation and for sharing your experience, it really helped clarify things for me.
I truly appreciate the time you dedicate not only to supporting users like me, but also to maintaining and improving this fantastic integration.
It's thanks to people like you that Home Assistant is such a powerful ecosystem.
I’ll go ahead with your suggestions, and if I need anything else, I’ll gladly reach out.
Thanks again!

closed 2025-05-23 by Caslinovich