stanus74/home-assistant-saj-h2-modbus #79: Support for ability to balance Battery Modules

odiechoo, 2025-06-08:
Hi. My SAJ configuration is 1x10kW inverter, coupled with 2 x 20kWh battery modules. So it should up as Battery 1 and Battery 2 in the system.

The Battery SOC that is reported in the Inverter, is the average of the SOC in Battery 1 and Battery 2. When I set my max battery charge depth (sensor.saj_battery_charge_depth) to 95% (to extend the life) via the SAJ App, I frequently get a large difference in SOC of Battery 1 (sensor.saj_battery_1_soc) and Battery 2 (sensor.saj_battery_2_soc). I'm currently on a 8% battery SOC difference between the 2, and have to go into the SAJ App to change it to 100% and have it charge to full to balance out the SOCs.

I would like to request, depending on what can be available, either:
a. registers that is able to set the max battery charge depth per battery bank. E.g. Max Depth for Battery 1 and Max Depth for Battery 2 ; or
b. If (a) is not possible, then the ability to write into the register for max battery charge depth, so that I can automate adjust it to 100% periodically, to have it balanced, and then return it back to 95%.

Thank you. I'm also open to any other potential solutions, not married to the above 2 options.

stanus74, 2025-06-08:
what I have found so far through tests is my assumption that the battery values SOC and others refer to the controller unit

I have 2 batteries myself and with e.g. Battery 2 Voltage, Temperature, Current ... I get a permanent value of 0 from the registers

Hence my recommendation to ask SAJ

Unfortunately SAJ do not want to support me, otherwise I could answer more questions

UDicke, 2025-06-08:
> Hi. My SAJ configuration is 1x10kW inverter, coupled with 2 x 20kWh battery modules. So it should up as Battery 1 and Battery 2 in the system.

Do the two 20 kWh batteries each consist of 4 x 5 kWh modules? I only know of the 5 kWh packs, which can be stacked to a maximum of 5 (25 kWh) to form a battery pack. That would explain the different understanding of battery 1 and 2, etc.

odiechoo, 2025-06-09:
Yes... Then that would be the Option (b) that I wrote in the original request. Where if we can write to the max charge depth for the whole battery set, then I can automate the balancing by setting the battery depth to 100% and checking the difference between individual SOC, before reverting it back to 95%.

odiechoo, 2025-06-10:
Oh.. I couldn't seem to find it... I did a search but it wasn't there. Will try again.

odiechoo, 2025-06-10:
Found it.... Thanks for showing. I was searching for "depth".. and scrolling through them all also somehow missed it.

Anyway, thanks. This will work sufficiently.

closed 2025-06-10 by odiechoo