stanus74/home-assistant-saj-h2-modbus #88: Problem after upgrade to HA Core 2025.8.0

jan120588, 2025-08-08:
Hello, after the update to HA 2025.8.0 all entities have the value “unknown”. Restart of plugin or HA didn't help.

[image]

UDicke, 2025-08-08:
I am running Version 2.3.0 of this integration and this is working fine with HA Core 2025.8.0 here.

stanus74, 2025-08-10:
Also did the HA update today and integration is working
Try deactivating, deleting, and reinstalling

jan120588, 2025-08-10:
Unfortunately, that doesn't work either. The following error appears in the HA log:

[image]

There was a similar problem with another integration:
https://github.com/pail23/stiebel_eltron_isg_component/issues/421

chatter80, 2025-08-10:
hy in Version 2025.4.2-beta2 is the integration working !

mrf68, 2025-08-10:
Just as UDicke: I am running Version 2.3.0 of this integration and this is working fine with HA Core 2025.8.0

stanus74, 2025-08-10:
Add or adjust this code at the end of configuration.yaml:

```
logger:
  default: warning  
  logs:    
    custom_components.saj_modbus: debug  
    homeassistant.components.modbus: error

```

Then post the contents of “/homeassistant/home-assistant.log”  after reboot

jan120588, 2025-08-10:
The following is repeated almost constantly:

2025-08-10 16:41:42.102 ERROR (MainThread) [custom_components.saj_h2_modbus.modbus_utils] Non-retriable exception occurred
Traceback (most recent call last):
  File "/config/custom_components/saj_h2_modbus/modbus_utils.py", line 48, in _retry_with_backoff
    return await func()
           ^^^^^^^^^^^^
  File "/config/custom_components/saj_h2_modbus/modbus_utils.py", line 182, in read_once
    response = await client.read_holding_registers(
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
2025-08-10 16:41:42.106 ERROR (MainThread) [custom_components.saj_h2_modbus.modbus_readers] Error reading inverter data: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
2025-08-10 16:41:42.106 ERROR (MainThread) [custom_components.saj_h2_modbus.modbus_utils] Non-retriable exception occurred
Traceback (most recent call last):
  File "/config/custom_components/saj_h2_modbus/modbus_utils.py", line 48, in _retry_with_backoff
    return await func()
           ^^^^^^^^^^^^
  File "/config/custom_components/saj_h2_modbus/modbus_utils.py", line 182, in read_once
    response = await client.read_holding_registers(
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^

closed 2025-08-17 by stanus74