stanus74/home-assistant-saj-h2-modbus #94: unexpected keyword argument 'slave'

seecode75, 2025-09-04:
### Inverter model

ASP 10KW-3P-X E2 

### Inverter firmware version

BMS-Software-Version	v1.30

### Which Modbus TCP converter are you using?

Other (please specify below)

### Modbus converter firmware version

?

### Additional converter details (if "Other" selected)

Ampere Solar Inverter (EKD-Solar)

### Describe the issue

The integration worked fine until the update HA 2025.8.3 was installed.

### Home Assistant version

 Core 2025.9.0   ;     Supervisor 2025.08.3 

### Log output

2025-09-04 13:05:27.267 WARNING (MainThread) [custom_components.saj_modbus.modbus_utils] Read attempt 1 for address 36608, unit 1: TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Traceback (most recent call last):
  File "/config/custom_components/saj_modbus/modbus_utils.py", line 70, in try_read_registers
    coro = client.read_holding_registers(address=address, count=count, slave=unit)
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'

frencifrenci, 2025-09-04:
Same problem after update HA Core to 2025.9.0

frencifrenci, 2025-09-05:
In file modbus_utils.py, edit row 130 to:

response = await client.read_holding_registers(address=address, count=count)

It seems to work

mr-gosh, 2025-09-05:
Same Problem after update to HA 2025.9.1

stanus74, 2025-09-05:
The adjustment was already made in version 2.3.1.
https://github.com/stanus74/home-assistant-saj-h2-modbus/commit/b0fb5fe535f464ea186b5fb161a1334d935772e7# diff-9b019b40a1a960dbe040729f687ce6e0533bfd0ad93b0362eb077a2a99465f93

Please ensure that you always have the latest version installed. 
https://github.com/stanus74/home-assistant-saj-h2-modbus/releases

seecode75, 2025-09-05:
In fact, I had used an old version. However, an update was not possible. After uninstalling, I was able to install the latest version. Now everything works perfectly again. Thank you and sorry for the inconvenience.