stanus74/home-assistant-saj-h2-modbus #95: [Bug/Support] - [Short Description] can"t read battery after updating to latest Home assistant

braham2019, 2025-09-06:
### Inverter model

AS2-5K-S-B

### Integration version

2.4.0

### Which Modbus TCP converter are you using?

AIO3

### Describe the issue

Can't get any data out of the battery since update to 2025.09. It has been working flawlessly for over a year.

### Home Assistant version

2025.9.1

### Installation type

Home Assistant OS

### Log output

This error originated from a custom integration.

Logger: custom_components.saj_modbus.modbus_readers
Source: custom_components/saj_modbus/modbus_readers.py:76
integration: SAJ H2 Modbus (documentation, issues)
First occurred: 14:25:14 (28 occurrences)
Last logged: 14:30:41

Error reading modbus data: Reconnection needed after 3 failed read attempts at address 13851, unit 1. Last error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Error reading modbus data: Reconnection needed after 3 failed read attempts at address 13914, unit 1. Last error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Error reading modbus data: Reconnection needed after 3 failed read attempts at address 13878, unit 1. Last error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Error reading modbus data: Reconnection needed after 3 failed read attempts at address 16388, unit 1. Last error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Error reading modbus data: Reconnection needed after 3 failed read attempts at address 16494, unit 1. Last error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'

This error originated from a custom integration.

Logger: custom_components.saj_modbus.hub
Source: custom_components/saj_modbus/hub.py:483
integration: SAJ H2 Modbus (documentation, issues)
First occurred: 14:26:42 (2 occurrences)
Last logged: 14:29:22

Error reading discharging state: Reconnection needed after 3 failed read attempts at address 13829, unit 1. Last error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'

### Pre-checks

- [x] I have updated to the latest release
- [x] I confirmed the inverter is reachable via Modbus TCP
- [x] I searched for existing issues

braham2019, 2025-09-06:
I removed the integration completely and deleted all the files. re-installed it and it is working fine now.

stanus74, 2025-09-13:
you use the wrong sensors
https://github.com/stanus74/home-assistant-saj-h2-modbus/discussions/73

stanus74, 2025-09-14:
**The daily sensors report daily values, and the total sensors report accumulated values — this has been the case since the very first version of the integration.**

If things no longer display correctly in the Home Assistant Energy Dashboard, it's most likely due to changes on the HA side — not the integration itself. In that case, your question would be better addressed in the Home Assistant forum.

stanus74, 2025-10-03:
make a new clean installation
https://github.com/stanus74/home-assistant-saj-h2-modbus/issues/94