stanus74/home-assistant-saj-h2-modbus #101: [Bug/Support] - Update missing sensor

Deapper, 2025-09-21:
### Inverter model

SAJ H2-10k

### Integration version

latest

### Which Modbus TCP converter are you using?

AIO3

### Describe the issue

Hi @stanus74 ,

Great update. Just in regards to the 10sec updates could you please add sensor.saj_ct_pv_power_watt as not everyone has solar direct but rather ac coupled

### Home Assistant version

latest

### Installation type

Home Assistant OS

### Log output

Not needed

### Pre-checks

- [x] I have updated to the latest release
- [x] I confirmed the inverter is reachable via Modbus TCP
- [x] I searched for existing issues

stanus74, 2025-09-26:
For fast sensors, the Modbus registers must also be read. Simply adding them to the HA Hub update, as you did, is not sufficient.

Perhaps CT Grid Power will be available in a future version. Since the smart meter registers are located in a different area, the effort involved is greater.

See new sensor in changelog.

[see new version changelog](https://github.com/stanus74/home-assistant-saj-h2-modbus/blob/main/CHANGELOG.md)

CT_PVPowerWatt does not exist according to the Modbus protocol. It was accidentally added in a previous version.

Unfortunately, SAJ does not support me with up-to-date documentation.

Deapper, 2025-09-26:
I don't really understand. I use CT_PVPowerWatt on my power flow card plus to display my solar feed in. Also adding the above code does make it update every 10 seconds. I have deleted the above code and confirmed it goes back to every 60 seconds which is what I have the integration set at. Are you saying you removed it in the latest update?

Deapper, 2025-09-26:
[image]

Deapper, 2025-09-27:
Ok I won't update thanks. 

Just to show you that the solar is updating every 10sec even when you said it shouldn't. Definitely not complaining and certainly don't want to offend you. 

https://youtu.be/cRYr0o1AVro

closed 2025-10-07 by stanus74